stm32-rs/fdcan #1: HAL support

richardeoin, 2021-11-11:
To make this crate useful and worth maintaining, it's helpful if it is used in at least a couple of HALs

- [ ] stm32g4xx-hal
- [ ] stm32h7xx-hal - [PR here](https://github.com/stm32-rs/stm32h7xx-hal/pull/277)
- [ ] stm32g0xx-hal

yorickdewid, 2021-12-03:
I see there is a PR for H7, we really need this. How can I participate?

richardeoin, 2021-12-04:
Hi! @yorickdewid This create is at an early stage of development, but you can already start experimenting by checking out this crate locally and using the [patch](https://doc.rust-lang.org/cargo/reference/overriding-dependencies.html) functionality in cargo. Bug fixes, usage reports and documentation are really appreciated!

In the `Cargo.toml` of your project:
```toml
[patch.crates-io]
stm32h7xx-hal = { git = "https://github.com/richardeoin/stm32h7xx-hal/tree/fdcan-fdcan" }
fdcan = { path = "../path/to/your/checkout/folder" }
```

yorickdewid, 2021-12-04:
I have the crate already running on the STMH7. So far no issues. However I have not tested the FD mode yet.

luctius, 2021-12-13:
PR for the stm32g4xx-hal crate here: [PR](https://github.com/stm32-rs/stm32g4xx-hal/pull/39).

kevswims, 2023-02-15:
Usage for the SMT32G4 HAL has been merged: https://github.com/stm32-rs/stm32g4xx-hal/pull/39