stm32-rs/fdcan #4: On STM32G4 ILE bits are switched

luctius, 2021-12-16:
I just noticed that on the stm32g4, the meanings of the ILE bits are switched. This is (by the reading of the datasheet) not the case on the H7.

[image]

(Read the fine print)

richardeoin, 2022-01-10:
That's really confusing. Have you been able to check this functionality on real stm32g4 hardware? I suppose its possible either the high-level (EINT1/EINT0) or detailed description (fdcan_intr0_it/fdcan_intr1_it) is correct

luctius, 2022-01-11:
Apologies for not being clear; I only found this after running into this issue with our hardware.

richardeoin, 2022-03-13:
Should be fixed in #12

closed 2023-01-30 by luctius

projectgus, 2024-09-05:
I ran into this as well on STM32G474, and I think I have gotten to the bottom of it:

As per @ho-ho-ho's comment, ILE EINT0 and EINT1 bits are not switched on STM32G4. i.e. EINT0==bit 0, EINT1==bit 1. The per-bit-value descriptions in the SVD file and the Reference Manual are wrong, but the per-bit descriptions and bit names in these places are correct. [ST's HAL driver](https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/master/Inc/stm32g4xx_hal_fdcan.h#L784) agrees with this, too.

In the stm32g4 device support crate until last month (and in the current release), the interrupt numbers for `FDCAN_INTR0_IT` and `FDCAN_INTR1_IT` **were** switched. See https://github.com/stm32-rs/stm32-rs/issues/456 They should be 21,22 but they are 22,21.

Therefore, currently:

* fdcan sets `fdcan::interrupt::InterruptLine::_0` and `fdcan::interrupt::InterruptLine::_1` backwards.
* If you use the current stm32g4 device support crate 0.15.1 then fdcan_intr0_it and fdcan_intr1_it are backwards.
* `fdcan::select_interrupt_line_1` has the correct polarity for the bits, but on g4 it produces incorrect behaviour for another reason - see #51 .

I think the correct course of action is to revert #12, and then either use the pre-release device support crate or manually swap `FDCAN_INTR0_IT` and `FDCAN_INTR1_IT` when implementing interrupt handlers.