stm32-rs/fdcan #9: Transmit and receive should use byte array

yorickdewid, 2022-02-17:
I think the `FdCan::transmit` and possibly the `FdCan::receive0` and `FdCan::receive1` should accept a byte array instead of a `u32` vector. I understand the use of `u32` for RAM alignment but I think this should be handled internally with an optional padding.

Public `FdCan` functions should not have to use lambda's if they are not necessary. This would make things more idiomatic Rust.

I can draft a PR.

yorickdewid, 2022-02-26:
@richardeoin What do you think?

luctius, 2022-02-26:
I'm all for it. I had my reasons for using u32, but you make a valid case for why it should have been a byte array.

richardeoin, 2022-02-27:
Agreed, I think PRs on either topic would be very welcome!

richardeoin, 2022-03-13:
Closed in #10

closed 2022-03-13 by richardeoin