stm32-rs/fdcan #11: Interrupts/registers do not match the H7 datasheet

yorickdewid, 2022-03-02:
The block reference manual (RM0433) lists different interrupts and registers than the current driver expects.

Found so far:
- RM044 section 56.5.16 `FDCAN_IE` is different than what [src/interrupt.rs](https://github.com/stm32-rs/fdcan/blob/master/src/interrupt.rs#L19) lists.
- RX FIFO (RXFnC.FnWM) watermark is never set which means it will default to 0. Any RAM index greater than 0 will be reported as 'overrun'.

richardeoin, 2022-03-13:
These are differences in the STM32H7 / STM32MP1 FDCAN controller that aren't captured by[ the feature gates](https://github.com/stm32-rs/fdcan/blob/master/Cargo.toml#L21-L22). Thanks for listing these two - have you found any more?

yorickdewid, 2022-03-13:
So for not, but I'm sure there are more.

richardeoin, 2022-03-14:
> RM044 section 56.5.16 FDCAN_IE is different than what [src/interrupt.rs](https://github.com/stm32-rs/fdcan/blob/master/src/interrupt.rs#L19) lists.

Fixed by #14 

> RX FIFO (RXFnC.FnWM) watermark is never set which means it will default to 0

I think that should be fixed in the H7 HAL implementation, specifically [here](https://github.com/stm32-rs/stm32h7xx-hal/pull/277/files#diff-8b316188eb9b5d776fbee9c54b421b8cdd716be51f21db9e8da618bfd1502f28R101)

David-OConnor, 2023-05-14:
So fun fact should anyone come across this, but on G4 you need to reverse the interrupt lines in both the NVIC name and when enabling the line; this error goes all the way back to ST's reference manual. (Works as expected on H7)