stm32-rs/fdcan #29: Request an example

David-OConnor, 2023-03-12:
Hi! Are there any examples available? Thoughts on adding one or more to an `examples` folder? I'm attempting to learn how to use this lib, but am not sure where to start. Thanks!

Of note, from the [docs page](https://docs.rs/fdcan/latest/fdcan/), it seems like most of the structs are for use in [Tx](https://docs.rs/fdcan/latest/fdcan/struct.Tx.html), [Rx](https://docs.rs/fdcan/latest/fdcan/struct.Rx.html), [FdCanControl](https://docs.rs/fdcan/latest/fdcan/struct.FdCanControl.html), with their methods being the primary API. Is this right?

richardeoin, 2023-03-12:
There are some examples available in downstream crates: [stm32g4xx-hal](https://github.com/stm32-rs/stm32g4xx-hal/blob/master/examples/can-echo.rs) [stm32h7xx-hal](https://github.com/stm32-rs/stm32h7xx-hal/blob/master/examples/can-fd.rs). It would be a nice improvement to have examples in this crate also.

Yes indeed, the methods on [FdCan](https://docs.rs/fdcan/latest/fdcan/struct.FdCan.html) are the "core" of the API.

David-OConnor, 2023-03-12:
Thanks! This looks like an outstanding lib. Checking those examples out.