stm32-rs/fdcan #39: can't get set_standard_filter to work

bjh-teleo, 2023-07-12:
Have tried this 
```
let arb_id = StandardId::new(0x700 as u16).unwrap().into();
let slot = StandardFilterSlot::_0;
let filter = StandardFilter {
    filter: FilterType::DedicatedSingle(ctlr_hb_arb_id),
    action: Action::StoreInFifo0,
};
can1.set_standard_filter(slot, filter);
```

and this
```
let filter = StandardFilter {
    filter: FilterType::BitMask {
         filter: 0x700,
         mask: 0x7ff,
    },
    action: Action::StoreInFifo0,
};
```

still receive all can messages

bjh-teleo, 2023-07-12:
Came across this https://github.com/stm32-rs/fdcan/issues/35, could be related.

David-OConnor, 2023-07-15:
Have you tried setting a reject all filter in a higher-numbered slot than your accept filter? That seems to be what I had to do from experimenting.

bjh-teleo, 2023-07-15:
No, that is a good idea! I will try that!

bjh-teleo, 2024-02-01:
@David-OConnor been a few months and I'm finally getting back to this. Can you share an example for how to set a "reject all" filter?

David-OConnor, 2024-02-01:
```rust
pub fn setup_protocol_filters(can: Can_) -> Can_ {
    let mut can = can.into_config_mode();

    // Node: H7 has up to 64 filters available. This is set up for G4's limitations.
    // This `GetNodeInfo` also matches dynamic ID allocation.

    let s = ServiceData {
        dest_node_id: 0, // 7 bits
        req_or_resp: RequestResponse::Request,
    };

    set_dronecan_filter(
        &mut can,
        ExtendedFilterSlot::_0,
        FrameType::Service(s),
        MsgType::GetNodeInfo.id(),
    );
    set_dronecan_filter(
        &mut can,
        ExtendedFilterSlot::_1,
        FrameType::Message,
        MsgType::IdAllocation.id(),
    );
    set_dronecan_filter(
        &mut can,
        ExtendedFilterSlot::_2,
        FrameType::Service(s),
        MsgType::GetSet.id(),
    );
    set_dronecan_filter(&mut can, ExtendedFilterSlot::_3, FrameType::Service(s), MsgType::Restart.id());

    // Place this reject filter in the filal slot, rejecting all messages not explicitly accepted
    // by our dronecan ID filters.
    let reject_filter = ExtendedFilter::reject_all();
    can.set_extended_filter(ExtendedFilterSlot::_7, reject_filter);

    can.into_normal()
}
```